DVC-Viking-Robotics/webapp #17: migrate dvcrobotics.tech to github pages.

2bndy5, 2019-09-05:
We can also use this outlet as a user's entry point for documentation. Also try to implement Bulma CSS framework simultaneously.

2bndy5, 2019-09-06:
@guidec We should probably start a separate repo for the dvcrobotics.tech site and just link to another repo's (ie webapp repo) webpage for docs & stuff specific to that repo. This way we can implement docs for just using the drivetrain code (will live in its own repo soon) and a different website for docs related to GPS code etc etc.

guidec, 2019-09-27:
Domain & Hosting is still active on Hostinger.com until "Expires 08/12/2021".That gives us a solid amount of time. 

It could happen that github pages has very slow page load times since the service is free.
I'll port all the files here, let's see how it performs.