DVC-Viking-Robotics/webapp #24: Connect sensor data to charts

tejashah88, 2019-09-16:
The sensor dashboard currently displays dummy data and should be ideally pulling the data from the robot via sockets.

2bndy5, 2019-09-29:
Looks like the call to trigger getting the sensor data via the socket `sensorDoF` (emitted from sockets.py) got lost or deleted during refactoring crusade or even earlier. We have to patch it back in. Currently, there's no websocket declaration on the sensors.html page. Also sensor-data.js was being used for getting both GPS and IMU data, but now its only good for GPS (the heading data is commented out).

tejashah88, 2019-09-29:
If you mean from `sockets.py`, it wasn't ever deleted: https://github.com/DVC-Viking-Robotics/webapp/blame/nondrive-R2D2/webapp/sockets.py#L147. As for the heading code, I commented it out for now until you had a stable implementation ready.

2bndy5, 2019-09-29:
not the file. the code in that file related to the socket called `sensorDoF`. And BTW @tejashah88 COMMENTS!!! I'm not sure how to inject the data into your charts. Additionally I think the `raw-data` element id should be renamed `imu-data` (or something a little more specific) since the raw data actually exists on the sensor only (raw data is without proper units; displayed data has conceivable units)

2bndy5, 2019-09-29:
IMU sensor data is getting to the `sensors.html` page, but not getting injected into the charts.

PR #56 (2019-10-06): Adding sensor integration to charts
Resolves #24. Adding more comments so do not merge yet!
closed 2019-10-06 by 2bndy5