DVC-Viking-Robotics/webapp #35: add a muxing class for multiple sensors of the same type

2bndy5, 2019-09-21:
I plan to have a GPS muxing class that takes into account the dilutions of precision attributes and differences of latitude and longitudes... Something similar should be devised for aggregating sensor data from multiple IMUs, but that will be a little more complex since each sensor device has a different set of capabilities (gyro + accel vs mag only vs gyro + accel + mag). 

To identify what type of IMU we're drawing from, we can use `if type(IMUs, LSM9DS1)` to isolate data from the 9 DoF sensor.

2bndy5, 2019-09-28:
@karleskinderic findings on [how to use sensor fusion](https://medium.com/@cotra.marko/wtf-is-sensor-fusion-part-1-laying-the-mathematical-foundation-89e2d304e23e)

karleskinderic, 2019-09-29:
I'll continue to do my research on what's possible...or at least understanding how different options would work and weighing the benefits of each.  I'll have to really more heavily on @2bndy5 when it comes to editing the actual code

2bndy5, 2019-09-30:
Throwing this out there... Maybe we just look at each sensor as an independent dataset from different parts of the robot.

karleskinderic, 2019-09-30:
I think that's always an option. We should at least test that and see if it
meets our purposes