DVC-Viking-Robotics/webapp #36: Which is really a required dependency anyway?

2bndy5, 2019-09-23:
we should be using a better way to ensure our direct dependencies and their sub-dependencies are cohesive. We can consider [this pipenv library](https://docs.pipenv.org/en/latest/) as an alternative/option to solve this issue

2bndy5, 2019-09-23:
I'm re-posting this from a Slack chat after slimming down the `requirements.txt` to only include libraries we are actually importing in our code with a special exception to eventlet per flask docs about production servers which are required for using websockets.

Most of the following links lead to the line in the repository's `setup.py` file that indicates what sub-dependencies they are installing during installation of our direct dependencies. Non-library related links lead to additional information.

- `git+<our-undeployed-repo-url>.git` commands require [git installed](https://git-scm.com/downloads) & added to the system's "PATH" environment variable (a commonly default condition on Linux, but not on Windows). Prefixing these items with `-e` makes importing modules from multi-module packages (like our Drivetrain library) much easier [using eggs](https://python101.pythonlibrary.org/chapter38_eggs.html#chapter-38-the-python-egg), but `pip` has trouble looking for the eggs during install. Most of our custom repository installs require:
            1.  adafruit-blinka (see link below)
            2. [adafruit-circuitpython-busdevice](https://github.com/adafruit/Adafruit_CircuitPython_BusDevice/blob/52d87bd7e571af66ecb57ee32b5af92531134150/setup.py#L37) (for I2C & SPI devices) 
            3. Our GPS_Serial requires pyserial (no python sub-dependencies, but can rely on `cp2110` and `CH340` USB serial drivers)

- [Flask](https://github.com/pallets/flask/blob/12666af6b5474998907310b2f940aebdc3b5a362/setup.py#L52)
- [Adafruit-Blinka](https://github.com/adafruit/Adafruit_Blinka/blob/bf49679028e57296e105c3e50b2cd25abc8dc19b/setup.py#L49)
- [Jinja2](https://github.com/pallets/jinja/blob/cc892563264562bc7ec3811dfa6a8ef9aa2960db/setup.py#L80)
- [Flask-SocketIO](https://github.com/miguelgrinberg/Flask-SocketIO/blob/f68e843422fd7989f071176aa5c8a1daabb6c14e/setup.py#L27)
- [python-socketio](https://github.com/miguelgrinberg/python-socketio/blob/a839a36fa0fa7f0e5d8976ff47b217f6b1e8a44b/setup.py#L31)
- [eventlet](https://github.com/eventlet/eventlet/blob/ac7917c7241bc29696cf65dd7b9757df9c9bc955/setup.py#L17)
- [opencv-python](https://github.com/skvark/opencv-python/blob/fd9fb99f2bdbe6e16c5bf1832bb8db5d0af65249/setup.py#L184)
- [pyxtermjs](https://github.com/cs01/pyxterm.js/blob/d7273ae79e231e6486faeeea4de9142faf21b256/setup.py#L63)

 These were added from _**sloppily**_ using `pip freeze`
- [autopep8](https://github.com/hhatto/autopep8/blob/120537a051d7f3cbbb5c4ede19b9e515156bd3d1/setup.py#L13)
- [pycodestyle (formerly called pep8)](https://github.com/PyCQA/pycodestyle/blob/9726e10004e5615c432f126530ee0540b9903f6e/setup.py#L38) has issues in this regard
- [pep8](https://github.com/treyhunner/pep8/blob/3c8abd0a91482a48ee47afbcf45d1b62e3340212/setup.py#L22)

Fortunately, we are no longer using [GPIOzero](https://github.com/gpiozero/gpiozero/blob/94d89bf414b1ce4dddaaa8be66d4684b0fcf4952/setup.py#L57)

As of this writing our requirements file now looks like this:
```

git+https://github.com/2bndy5/Adafruit_CircuitPython_LSM9DS1.git
git+https://github.com/2bndy5/CircuitPython_MPU6050.git
git+https://github.com/DVC-Viking-Robotics/Drivetrain.git
git+https://github.com/DVC-Viking-Robotics/GPS_Serial.git
circuitpython-nrf24l01==1.1.0
opencv-python==4.1.1.26
pyserial==3.4
Flask==1.0.2
Flask-CacheBuster==1.0.0
Flask-Compress==1.4.0
Flask-Login==0.4.1
Flask-SocketIO==3.3.2
eventlet==0.24.1
pyxtermjs==0.4.0.1

```

As a reminder, you can use `pip list` to have `pip` spit out a list of all packages installed in your current environments site-packages folder and elsewhere if you added to python's "PATH" environment variable.

tejashah88, 2019-10-05:
After doing some investigating with the Pipenv project, here's what I found:
* Pipenv uses the concept of **lock files**, which are comprehensive documents that specify exactly what dependency versions and sub-dependency versions should be installed, right down to the sha256 values. This also has the added benefit of security, should the network be compromised.
* The `requirements.txt` would be replaced by the new `Pipfile`, which not only specifies the dependency versions but allows you to specify developer dependencies and the python version needed for the project. This is similar to how npm works, with their corresponding `package.json` and `package-lock.json`.
* Upon running `pipenv install`, it will automatically create a virtual environment that sits outside of the project folder (for the case of Ubuntu: `~/.local/share/virtualenvs/`) as well as install the dependencies from the `Pipfile` and `Pipfile.lock` files.
* The installations via pipenv are designed to be deterministic (meaning we **always** get the same version for every single library), which is exactly what we want to solve this problem. This also means that the way of installing dependencies will be different from the native python way.

tejashah88, 2019-10-05:
Here's a list of commands for using pipenv, which is mostly coming from [this website](https://pipenv.kennethreitz.org/en/latest/basics/):

## Installing pipenv
```bash
pip install --user pipenv
```

## Setup
Create a new virtual environment:
```bash
pipenv --python 3.6  # Any *installed* python version works
```

Removing said virtual environment:
```bash
pipenv --rm
```

You can install the dependencies if there is an existing `Pipfile`:
```bash
pipenv install  # you can also add the '--dev' flag to install the dev dependencies
```

If you want to ensure that the dependencies are installed exactly from `Pipfile.lock`, you can run the following:
```bash
pipenv sync
```

In the case that a `Pipfile` doesn't exist, you can create one after installing *at least* one package:
```bash
pipenv install <package>
```

You can also import a `requirements.txt` file to create your `Pipenv` file from there:
```bash
pipenv install -r path/to/requirements.txt
```

Generating a `Pipenv.lock` lock file is as follows:
```bash
pipenv lock # you can include pre-released with the '--pre' flag
```

## Installing/Uninstalling
You can install or uninstall any package with the following commands (similar to pip):
```bash
pipenv install <package>
pipenv uninstall <package>
```

You can also install specific or semi-specific versions with the following modification:
```bash
pipenv install "requests~=1.2"
pipenv install "requests>=1.4"
```
You can find more info about this [here](https://pipenv.kennethreitz.org/en/latest/basics/#specifying-versions-of-a-package).

If you need to nuke your virtual environment and start from scratch, you can run the following command. **NOTE: THIS IS A DESTRUCTIVE ACTION**
```bash
pipenv uninstall --all
```

## Checking dependencies
You can check your dependencies for security vulnerabilities:
```bash
pipenv check
```

You can also see a dependency tree of your project:
```bash
pipenv graph
```

## Running the project
Since we don't have easy access to the virtual environment folder, the procedure to running the project has changed slightly. There's two ways to do so:
1. Activate the virtual environment and then run the project normally:
```bash
pipenv shell
python -m webapp.app
```
2. Run a command directly in the virtual environment without explicitly activating the env itself:
```bash
pipenv run python -m webapp.app
```

Note: You can run any command in the virtual environment via `pipenv run [command]`

tejashah88, 2019-10-05:
Here's the output of the Pipenv file (as of 10/5/2019):
```TOML
[[source]]
name = "pypi"
url = "https://pypi.org/simple"
verify_ssl = true

[dev-packages]

[packages]
adafruit-circuitpython-lsm9ds1 = {git = "https://github.com/2bndy5/Adafruit_CircuitPython_LSM9DS1.git"}
circuitpython-mpu6050 = {git = "https://github.com/2bndy5/CircuitPython_MPU6050.git"}
drivetrain = {git = "https://github.com/DVC-Viking-Robotics/Drivetrain.git"}
gps-serial = {git = "https://github.com/DVC-Viking-Robotics/GPS_Serial.git"}
circuitpython-nrf24l01 = "==1.1.0"
opencv-python = "==3.4.3.18"
pyserial = "==3.4"
eventlet = "==0.24.1"
pyxtermjs = "==0.4.0.1"
Flask = "==1.0.2"
Flask-CacheBuster = "==1.0.0"
Flask-Compress = "==1.4.0"
Flask-Login = "==0.4.1"
Flask-SocketIO = "==3.3.2"

[requires]
python_version = "3.6"
```

tejashah88, 2019-10-05:
Here's the output of the Pipenv lock file (as of 10/5/2019):
```JSON
{
    "_meta": {
        "hash": {
            "sha256": "1406952e0b7101ee80fe107c40426acd64aef701ac6ed6fbce506303eb2afc70"
        },
        "pipfile-spec": 6,
        "requires": {
            "python_version": "3.6"
        },
        "sources": [
            {
                "name": "pypi",
                "url": "https://pypi.org/simple",
                "verify_ssl": true
            }
        ]
    },
    "default": {
        "adafruit-blinka": {
            "hashes": [
                "sha256:805dfa44ee486ee4903233f4c40a41605cd63cc86190de47cbdb45e4c4207961"
            ],
            "version": "==2.5.3"
        },
        "adafruit-circuitpython-busdevice": {
            "hashes": [
                "sha256:07a1430946b73925f98c6251b972bdf52c620d880150b1b4f5c7eff85b05fae7"
            ],
            "version": "==4.0.0"
        },
        "adafruit-circuitpython-lsm9ds1": {
            "git": "https://github.com/2bndy5/Adafruit_CircuitPython_LSM9DS1.git",
            "ref": "96282bba8809cf08a3c4fd85f89146bc471f2a0a"
        },
        "adafruit-platformdetect": {
            "hashes": [
                "sha256:3ea62dad76efb1fee9004fefbe392274c8e86b565cef72ca5fbf768459ee0637"
            ],
            "version": "==1.3.4"
        },
        "adafruit-pureio": {
            "hashes": [
                "sha256:e65cd929f1d8e109513ed1e457c2742bf4f15349c1a9b7f5b1e04191624d7488"
            ],
            "version": "==0.2.3"
        },
        "circuitpython-mpu6050": {
            "git": "https://github.com/2bndy5/CircuitPython_MPU6050.git",
            "ref": "708ace623887e70071b55fe49697c9745c89e5ae"
        },
        "circuitpython-nrf24l01": {
            "hashes": [
                "sha256:91d65464b99ab09c64b1de3de3a6226c256af199e5ae9216ecbff24844a46b38"
            ],
            "index": "pypi",
            "version": "==1.1.0"
        },
        "click": {
            "hashes": [
                "sha256:2335065e6395b9e67ca716de5f7526736bfa6ceead690adf616d925bdc622b13",
                "sha256:5b94b49521f6456670fdb30cd82a4eca9412788a93fa6dd6df72c94d5a8ff2d7"
            ],
            "version": "==7.0"
        },
        "dnspython": {
            "hashes": [
                "sha256:36c5e8e38d4369a08b6780b7f27d790a292b2b08eea01607865bf0936c558e01",
                "sha256:f69c21288a962f4da86e56c4905b49d11aba7938d3d740e80d9e366ee4f1632d"
[... 225 lines cut ...]
        "spidev": {
            "hashes": [
                "sha256:4314e52f573d95233c907f307558893313a8a606e197e77bb711526b0e179e80"
            ],
            "version": "==3.4"
        },
        "sysv-ipc": {
            "hashes": [
                "sha256:b284e961274c67ed3debfdd8a43f197451c5f100955142bbed380258c6c73e06",
                "sha256:b59fbd42c9de27b42fdda69c4aa7556312112dc6ac7246e44cd875cf049e4273",
                "sha256:cd70548a92218ede2da499080b31e52f1d7cf60716c5cc7d468ec89e8b73c530"
            ],
            "version": "==1.0.0"
        },
        "werkzeug": {
            "hashes": [
                "sha256:7280924747b5733b246fe23972186c6b348f9ae29724135a6dfc1e53cea433e7",
                "sha256:e5f4a1f98b52b18a93da705a7458e55afb26f32bff83ff5d19189f92462d65c4"
            ],
            "version": "==0.16.0"
        }
    },
    "develop": {}
}
```