DVC-Viking-Robotics/webapp #41: create a new config wizard/walkthrough

2bndy5, 2019-09-30:
There needs to be a first-run wizard to gather information about different configurable aspects for both hardware and user preferences. Hardware configuration is being saved to a file named HWconfig.json, and user-specific stuff should go into a file with a title that incorporates the username of the account.

tejashah88, 2019-10-16:
Would this be a CLI utility or part of the web app itself?

2bndy5, 2019-10-16:
Ideally part of the webapp, but it could be done in the backend after the websocket clients (with all the hardware connected) has established a connection. This issue is kinda up in the air till the separation of hardware is finalized.