DVC-Viking-Robotics/webapp #42: [Web design] Improve web app responsiveness

tejashah88, 2019-09-30:
Some of the web pages aren't optimized for mobile use and could use some CSS fine-tuning. 

Here's what the Autonomous Mode page shows: 
[image]

tejashah88, 2019-10-01:
FYI, the other pages display fairly well with the mobile view

2bndy5, 2019-10-01:
we should comment out [the sidebar](https://github.com/DVC-Viking-Robotics/webapp/blob/0cc6ff8c4ca211721e79083b4c9b12189f1f5a81/webapp/templates/layout.html#L56-L78) and [its hamburger button](https://github.com/DVC-Viking-Robotics/webapp/blob/0cc6ff8c4ca211721e79083b4c9b12189f1f5a81/webapp/templates/layout.html#L84-L86) until the user accounts/preferences and the remotes are re-done

tejashah88, 2019-10-01:
Opened issue #45 for that. Until then, this issue has been addressed

closed 2019-10-01 by tejashah88